ainfosec/FISSURE #12: Current GNURadio Version Support

muaddib1984, 2022-08-26:
Will you be porting this to the current version of GNURadio (3.10)?

cpoore1, 2022-08-26:
That's a good question and I'm actively thinking about the best way to go about it. The current plan is to use it for Ubuntu 22.04 so I have somewhere to go that does have 3.10. I know I'm not getting the most out of GNU Radio by using the older versions (3.7/3.8), but there are so many tools that exist only for the older versions. If I support multiple types of GNU Radio in one operating system it will probably become a mess that won't make it intuitive for everyone to navigate on their own. I could make it an option and install only 3.10 from source for something like 20.04. It wouldn't be too much of a hassle since it would share the same FISSURE branch as 22.04 - I could just set a flag to use a different collection of flow graphs.

Unfortunately, I don't know much about 3.10 as of today. I haven't successfully converted an OOT module yet and it is the main reason why FISSURE isn't really available for Ubuntu 22.04. That and I'm still trying to troubleshoot the installation for a bunch of programs in my list. I will get to it but I need a little time to get up to speed.

I'm open to thoughts and suggestions from anyone on this topic.

muaddib1984, 2022-08-26:
Thanks for the response. WRT old blocks, [bkerler](https://github.com/bkerler/) has ported nearly 60 GNURadio blocks that were abandoned or had 'gone stale'. You can see them in his build script [here](https://github.com/bkerler/gnuradio_install/blob/main/prepare.sh). I've been speaking with him on chat.gnuradio.org and he offered to try and port FISSURE. I'd suggest you hop on there and start a dialog with him as he's spun up on porting old blocks.
Two important reasons for updating to 3.10 are:
1)  GNURadio 3.10 is pinned as the final 3.X release and will have solid support for the foreseeable future, so it's a good one to latch onto. It's also a much better version that 3.8 IMO (I've been using GNURadio since 3.6). So it won't be a treadmill of staying up-to-date, until you do a 4.0 version (which is a good ways out) 
2) The GNURadio developers really hustled (and succeeded) in getting GNURadio 3.10 included in the Jammy Repo. So ```apt install gnuradio``` in Jammy will give you 3.10. They've also put massive efforts into the GNURadio PPA if you want the latest 3.10.  

Just being frank, your project seems neat, but i'm not going to rollback GNURadio to use it. I mean that with the utmost respect, but I use it everyday and I like the way it's developing. I'm not a 'docker guy' but I've been jumping back in solely to try and install your project in an isolated environment. I have a PyBombs 3.8 env, but I don't think your install script will respect the paths there (and i'm not going to risk it). So If you could provide a Dockerfile for FISSURE, that would also be a great 'try-before-you-buy' option. Apologies if i'm missing a Dockerfile, but I didn't see one in your source tree.

cpoore1, 2022-08-26:
This is all great information. I'll keep in touch with bkerler and try to get 3.10 going in the next couple days (Ubuntu 22.04 as a starting point). We've been discussing a docker option on our end because I know all this software is a lot for anyone and it doesn't play nice with what's already there. I imagine a lot of people share the same thoughts as you and I'll try to make this useful to as many people as I can. It helps me the more I hear about the different setups people like to use.

muaddib1984, 2022-08-26:
For sure. Thanks for being receptive. Bkerler has already porting your project :)

alphafox02, 2022-08-27:
3.10 will be great, but continued 3.8 support would also be great (if possible). Beyond OOT modules, that still some great applications that build upon 3.8 to include dji droneid, Boatbod op25 and Qradiolink. I was quite pleased to see you all had taken the time to release a 3.8 based app and look forward to trying a 3.10 version in the future.

muaddib1984, 2022-09-03:
@cpoore1 bkerler has now ported ALL blocks needed for fissure to 3.10. Including gr-Bluetooth (which many have tried and failed). A 3.10 port should only require his blocks be added to the architecture. Hard part is done :)