ainfosec/FISSURE #13: Explain what RF is

89z, 2022-08-28:
It seems RF as used in this repo is radio frequency:

https://wikipedia.org/wiki/Radio_frequency

however the Readme doesn't says the phrase radio frequency a single time. I think it would be helpful to add this phrase at least once, so a broad audience knows the purpose of this repo.

cpoore1, 2022-08-28:
Sure, we can do that. With a little time and some input from the community we can develop helpful guides within FISSURE and point people towards them (even from the README). The access to all these tools and visualizations makes this software a great candidate for teaching others about several types of technology. Still a lot of work to be done. 

This goes for everyone: please add new items in the Discussion tab with suggestions on lesson material or ways to present things differently. That's the main way to communicate for now.

cpoore1, 2022-09-04:
I'll also add more information in the weeks to come about the principles behind an RF framework, how well FISSURE currently meets that criteria, and what still needs to be realized. It will likely be a link to a different page with more images and videos.