ainfosec/FISSURE #16: Can't install (KDE Neon)

Foul, 2022-08-29:
hello !

I'm on KDE Neon (based on Ubuntu 20.04 Focal) and when trying to install
./install

This branch is for Ubuntu 20.04 and 22.04 only. Switch to a different FISSURE branch before installing!

thx.

cpoore1, 2022-08-29:
I haven’t checked KDE Neon out yet but it must be failing the initial check in the “install” file (root directory). You can modify the check or just run the 20.04 commands listed in that file. Then run the second “installer” file in the Installer folder. You should be able to manually pick whatever 20.04 version after that and hope for the best.

Foul, 2022-08-29:
i successfully runned the second installer ;) thx !

cpoore1, 2022-09-08:
I added KDE neon 5.25 to the installer. It repeats all the steps for 20.04.4. I saw that it asks you to upgrade to 22.04 but you can't download it yet from the website. Not sure what's up with that. It looks like it all installed fine for me with the exception of the OOTs showing up as red the first time using the installer when they should be green. I'll close this issue once I figure out a good way of dealing with that.

closed 2022-10-30 by cpoore1