ainfosec/FISSURE #19: Way to install on Debian 10

nico0481, 2022-09-01:
Hello,

Is there a way to install Fissure on Debian 10 ?

Thanks

cpoore1, 2022-09-01:
Not yet, but anyone is welcome to help me get a head start by modifying the installer. A partial list of ways to install some of the programs would speed things up for me. I’ll start to look into it though.

0ofta, 2022-09-05:
Hi, 

I am currently adding Arch Linux to the installer, and for now,  the minimal install is running without problems, but there is a lot of software to adapt, so it takes time.

When I am finished with this, I could look at debian, if no-one else has gotton around to it yet :)

@cpoore1: Do you want a separate branch specifically to Arch, and Debian - or should the installer be able to install all distros (from the default branch), that have been adapted?

cpoore1, 2022-09-05:
Awesome. I prefer to stick to the two branches and make adaptations in the installer and code for special cases. It is less work for me when updating small changes across all the branches. If it is too different I can rethink that.

nico0481, 2022-09-06:
> Not yet, but anyone is welcome to help me get a head start by modifying the installer. A partial list of ways to install some of the programs would speed things up for me. I’ll start to look into it though.

Hi, Fissure seems to be a very great tool.
I would like to contribute, but unfortunately I've no sufficient skills in Linux to help.

Thank you
Regards

cpoore1, 2022-09-06:
@nico0481 There are a lot of ways to contribute. If you want to talk more about it, open a new discussion in the Discussions tab and we can think of something. It would benefit others thinking the same thing if we moved the conversation over to there.

wernight, 2022-11-03:
It almost installs flawlessly on Debian buster. At least the 1.0.1-Python-3.8.

The main issue I have is that it really wants to have `gnome-terminal` which isn't the terminal I have.

cpoore1, 2022-11-06:
I'm not sure of the best way to replace gnome-terminal. I use it to open terminals and run programs or in conjunction with expect to open a terminal and display a command without running it. I don't really know how to get something like `x-terminal-emulator` to do anything. I could make a "default terminal" option in the settings if I had a list of terminal command that can do the same behavior as gnome-terminal.