ainfosec/FISSURE #21: Fissure wont start after sucesful install

porlkertus, 2022-09-01:
Excited to try the tools but sadly the main fissure application wont run from the UI or Command Line after a seemingly sucessful install. Running latest Ubuntu and a very recent install aprt from GNURadio and RFCat

cpoore1, 2022-09-01:
FISSURE for Ubuntu 22.04 will be updated and tested within the next few days. The major holdups were the missing 3.10 OOT modules but they should be ready soon. If the Dashboard/GUI from the `fissure` command doesn’t load then one of the Python3 imports likely failed. Try opening a terminal and import the lines from the file throwing the error in `python3` until you find the program that’s causing it.

cpoore1, 2023-04-24:
This was a long time ago and before the Ubuntu 22.04 branch was tested. Not a lot of specific details to go on so I'm closing this issue for now.

closed 2023-04-24 by cpoore1