ainfosec/FISSURE #22: Ubuntu 22.04 Install Dependencies

kaiyazeera, 2022-09-02:
In order to be able to follow the installation instructions on a somewhat fresh Ubuntu installation, I was required to install the following packages:

```
sudo apt install build-essential git cmake libsndfile1-dev libjack-jackd2-dev libfftw3-dev emacs libudev-dev
sudo apt install qtbase5-dev qt5-qmake qttools5-dev qttools5-dev-tools qtwebengine5-dev libqt5svg5-dev libqt5websockets5-dev
```
Note that the list might contain additional packages that are not required or already installed per default.
Consider adding the dependencies to the Ubuntu 22.04 installation instructions.

kaiyazeera, 2022-09-05:
I did a full install (or at least tried) since I had none of the tools already installed.
The installer threw warnings that QT5 wasn't installed and aborted right at the start.
I did not however save the output so I cannot provide more detail (also out of a lack of time).

cpoore1, 2022-09-05:
Did you first run `./install` from the root directory or did you go right to the file in the Installer folder? There should be a prompt in the "install" file saying Qt5 isn't installed and ask you to install it (I added the prompt recently). If you did do that and it skipped over I might not be checking for the right things. Anyone else can leave a comment if they are seeing this too.

Regardless, support for 22.04 is still in the works. I'm figuring out the bugs in the installer for many of the tools and I need to test a whole set of 3.10 flow graphs for the Dashboard to use. I'll update the README when I feel good about it. In the meantime, you can still launch (some) tools from the menu and do a subset of the Dashboard functionality.

kaiyazeera, 2022-09-05:
I performed `./install` on the root directory. 
There was no prompt, however, only the stdout errors from which I assumed QT5 was missing. Thank you very much for the effort :)