ainfosec/FISSURE #32: Does it work on Kali?

nico0481, 2023-04-04:
Hello, 
Do you think it will work on Kali?
Or is this planned? 

Thanks

cpoore1, 2023-04-04:
I think it will work eventually on Kali. I’ve been waiting for someone to ask about it and haven’t tried going through it yet. I’ll get the latest version and take a look.

nico0481, 2023-04-04:
But which option to select that best matches the operating system in the installer?

cpoore1, 2023-04-04:
The Python3_maint-3.10 branch is what I’ll look at first. The Ubuntu 22.04 option is closest. I’ll have to change the hardcoded gnuradio version in the installer and avoid installing the 20 or so programs that already exist in Kali. Then change how they get called in the menu items. The minimal install will likely be close to working.

cpoore1, 2023-04-05:
I loaded up a fresh Kali 23.1 with the default settings and it looks like you are going to have some trouble. There are some pretty obvious GUI issues going on. You might have better luck with selecting GNOME in the install. I'll update you when I get a fix for it.

cpoore1, 2023-04-09:
Many of the GUI issues for the installer and the FISSURE light theme should be fixed, so now you can at least look at FISSURE in Kali. I added Kali to the installer and tried to uncheck the programs by default that already exist or failed during installation. There are still GUI elements that do not look right and I imagine there are going to be a lot of errors popping up so be cautious when installing and using things. I'll remove the beta designation in the installer and README when I think I cleaned up most of it.

nico0481, 2024-10-02:
I'm now trying to install on Kali (Kali Rolling 2024.3, Python 3.12.6)
After long installation with some non-blocking warning, when complete, i get
$ fissure
fissure: command not found

cpoore1, 2024-10-02:
If the fissure command was installed from the checkbox in the installer it should be ok. You might have to open a new terminal to try again. I'll see if I can refresh that.

rabarar, 2024-10-02:
That’s some lede, via @politico:

@kyledcheney 
https://t.co/1PloqIT1PgCarl Quintanilla ***@***.***)x.comThoughts?

nico0481, 2024-10-02:
It was installed with installer GUI,  with auto selected kali os and default check boxes

cpoore1, 2024-10-03:
Looks like this is a new Kali version I haven't tried yet. I'll take a look. If the fissure command doesn't work you can try executing the commands that are listed when clicking the check box that cd to that ./fissure directory, add it to the PYTHONPATH, and run the ./Dashboard/__main__.py file.

nico0481, 2024-10-03:
I get this answer:
python ./Dashboard/__main__.py                                        
Traceback (most recent call last):
  File "/home/nico/Documents/FISSURE/fissure/./Dashboard/__main__.py", line 2, in <module>
    from fissure.Dashboard.Frontend import Dashboard
ModuleNotFoundError: No module named 'fissure'

cpoore1, 2024-10-03:
That's what the export line does for that checkbox. For some reason that ./fissure folder isn't being recognized on its own with all the `__init__.py` files. It will temporarily get your open terminal to recognize it. With your user name and FISSURE location:
```
export PYTHONPATH=$PYTHONPATH:/home/user/FISSURE
cd /home/user/FISSURE
python3 /home/user/FISSURE/fissure/Dashboard/__main__.py
```
I guess the cd isn't really necessary but I tend to leave some unintentional artifacts sometimes so it's nice to know where they go. I think I intended to do `python3 fissure/Dashboard/__main__.py` but it works the same.

nico0481, 2024-10-07:
I dont't understand what you mean exactly, sorry

cpoore1, 2024-10-07:
I tried it on the latest Kali and there some issues that I'll try to fix up in the next day or two. I need to fix all the pip commands by putting `--break-system-packages` at the end (temporary fix until we start to use virtual environments) and put the rest of the installer in light mode so the all the widgets are visible. I'll see how the rest of the programs in the installer are faring and update the disk usage for everything.

cpoore1, 2024-10-19:
I think I fixed most of the things wrong with Kali 2024.3. I thought I broke the menu items but it takes like 10 seconds for it populate in a terminal for some reason. It was doing that for earlier Kali versions but I forgot about it.

closed 2025-10-16 by cpoore1