ainfosec/FISSURE #35: Fixed_threshold_b2x0.py does not work in 3.10 branch

seanee, 2023-05-08:
Hi, I've found that _fixed_threshold_b2x0.py_  is broken. It tries to import GrRangeWidget from gnuradio.qtgui but GrRangeWidget does not exist (that I can see) in qtgui.  I'm not sure if this is a version or something odd in my environment but I've checked that I am running the correct version of GNU radio and the most recent version of FISSURE. Has anyone seen this or had problems calling GrRangeWidget from other device types? 

Here is the error: 
"
TSI: Starting TSI Detector...
Traceback (most recent call last):
  File "/home/johndoe/FISSURE/Flow Graph Library/TSI Flow Graphs/fixed_threshold_b2x0.py", line 41, in <module>
    from gnuradio.qtgui import Range, GrRangeWidget
ImportError: cannot import name 'GrRangeWidget' from 'gnuradio.qtgui' (/usr/lib/python3/dist-packages/gnuradio/qtgui/__init__.py)
"

cpoore1, 2023-05-08:
What version of GNU Radio are you using and what happens if you drop a QT GUI Range block in GNU Radio Companion?
1. `gnuradio-companion`
2. Get version in the Console Panel
3. Ctrl+F and type "Range"
4. Drop block on any flow graph

If that works fine for version 3.10, you can try running the flow graph manually or rebuilding it. It should pop up a window and run from GNU Radio companion regardless of what FISSURE is doing.

cpoore1, 2023-05-10:
I ran through it on a fresh install and you are correct. I get the same error but it goes away after running grcc. I will add an extra step to the installer for all the FISSURE flow graphs. I may hold off on other people’s OOT modules until I can test them out.

cpoore1, 2023-05-14:
I added a "Compile Flow Graphs" option in the installer. It should take around 15 minutes to do everything. I will try to keep the FISSURE flow graphs compiled and updated to match the GNU Radio version that gets installed going forward. This command in the "/FISSURE/Flow Graph Library" folder will do the same thing: `find . -name '*.grc' -exec grcc {} \;`