ainfosec/FISSURE #45: FISSURE install on newly created Ubuntu 22.04 on WSL on Windows 10

peterdri, 2023-08-17:
installation on newly created Ubuntu 22.04 on WSL on Windows 10

can anyone help?

Thanks
Peter

eterd@DESKTOP-MGMJK4K:~/FISSURE$ ./install
Ubuntu 22.04 detected
./install: line 55: qmake: command not found
No Qt5 found.
Qt5 is required to open the installer. Do you want to proceed? [y]/n: y
[sudo] password for peterd:
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python3-pyqt5 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Unable to locate package qtcreator
E: Unable to locate package qtbase5-dev
E: Unable to locate package qtchooser
E: Unable to locate package qt5-qmake
E: Unable to locate package qtbase5-dev-tools
E: Package 'python3-pyqt5' has no installation candidate
Traceback (most recent call last):
  File "/home/peterd/FISSURE/Installer/installer", line 2, in <module>
    from PyQt5 import QtCore, QtGui, uic, QtWidgets
ModuleNotFoundError: No module named 'PyQt5'
peterd@DESKTOP-MGMJK4K:~/FISSURE$ pwd
/home/peterd/FISSURE

cpoore1, 2023-08-17:
I haven't tried that exact setup yet, but Ubuntu 22.04 should have the python3-pyqt5. Try doing a `sudo apt-get update` first. Also make sure you have apt working with an internet connection. What's the output of `lsb_release -d`?

peterdri, 2023-08-17:
Thanks for your comment.   Here is the result

peterd@DESKTOP-MGMJK4K:~/FISSURE$ lsb_release -d
Description:    Ubuntu 22.04.2 LTS
peterd@DESKTOP-MGMJK4K:~/FISSURE$ ./install
Ubuntu 22.04 detected
./install: line 55: qmake: command not found
No Qt5 found.
Qt5 is required to open the installer. Do you want to proceed? [y]/n: n
peterd@DESKTOP-MGMJK4K:~/FISSURE$

peterdri, 2023-08-17:
tried on different Ubuntu versions

peterd@DESKTOP-MGMJK4K:~/FISSURE$ ./install
Ubuntu 20.04 detected
./install: line 58: qmake: command not found
No Qt5 found.
Qt5 is required to open the installer. Do you want to proceed? [y]/n: n
peterd@DESKTOP-MGMJK4K:~/FISSURE$

peterd@DESKTOP-MGMJK4K:~/FISSURE$ ./install
Ubuntu 18.04 detected
No Python!
Python2 is required to open the installer. Do you want to proceed? [y]/n: n
peterd@DESKTOP-MGMJK4K:~/FISSURE$

peterdri, 2023-08-17:
after installing Python2 on 18.04

peterd@DESKTOP-MGMJK4K:~/FISSURE$ ./install
Ubuntu 18.04 detected
No Qt4!
Qt4 is required to open the installer. Do you want to proceed? [y]/n: n
peterd@DESKTOP-MGMJK4K:~/FISSURE$

cpoore1, 2023-08-17:
It errors for those too when you select yes? Choosing no exits out.

peterdri, 2023-08-17:
I did

sudo apt install qt4-default

then did ./install and selected yes and it is installing now.  Got to a GUI where I see a list of software such as LTE

cpoore1, 2023-08-18:
There could be a conflict with some other software or you might be missing the universe ppa which is normally a default ppa for Ubuntu. You could check with:
```
apt-cache policy python3-pyqt5
cat /etc/apt/sources.list
sudo grep -rhE ^deb /etc/apt/sources.list*
sudo apt-cache policy
apt policy
```
To add it:
```
sudo add-apt-repository universe
sudo apt-get update
```
To see if some other program is influencing it, you might be able to get something from a reverse search:
```
apt-cache showpkg python3-pyqt5
```
Then purge and reinstall the offending program.

peterdri, 2023-08-18:
The installation completed.   Then on the GUI I accidentally hit install instead of close.   I interrupted it quickly as shown below.
However fissure does not run.. Did I ruin the installation after 24 hours?

Thanks

[100%] Built target wavingz-test
/home/peterd/Installed_by_FISSURE/waving-z/build/wave-in
VERIFY SUCCESS

Install Complete
Hit:1 http://dl.google.com/linux/earth/deb stable InRelease
Hit:2 http://ppa.launchpad.net/nemh/systemback/ubuntu xenial InRelease
Hit:3 http://security.ubuntu.com/ubuntu bionic-security InRelease
Hit:4 http://archive.ubuntu.com/ubuntu bionic InRelease
Hit:5 http://downloads.metasploit.com/data/releases/metasploit-framework/apt lucid InRelease
Hit:6 http://archive.ubuntu.com/ubuntu bionic-updates InRelease
Hit:7 http://ppa.launchpad.net/git-core/ppa/ubuntu bionic InRelease
Hit:8 http://archive.ubuntu.com/ubuntu bionic-backports InRelease
Hit:9 https://openhab.jfrog.io/artifactory/openhab-linuxpkg stable InRelease
Hit:10 http://ppa.launchpad.net/gnuradio/gnuradio-releases-3.7/ubuntu bionic InRelease
Hit:11 http://ppa.launchpad.net/mkusb/ppa/ubuntu bionic InRelease
Hit:12 http://ppa.launchpad.net/myriadrf/drivers/ubuntu bionic InRelease
Hit:13 http://ppa.launchpad.net/wireshark-dev/stable/ubuntu bionic InRelease
0% [10 InRelease gpgv 15.9 kB]^CTraceback (most recent call last):
  File "/home/peterd/FISSURE/Installer/installer", line 5888, in _slotOK_Clicked
    time.sleep(0.1)
KeyboardInterrupt
peterd@DESKTOP-MGMJK4K:~/FISSURE$ fissure
fissure: command not found

cpoore1, 2023-08-18:
You may have to open a new terminal for the fissure command to work. You should be ok if you canceled it immediately. There are only a couple things that can’t be run twice and I don’t think it would be that bad if they did. You can see what commands get run if you are concerned and try to fix anything being written twice.

peterdri, 2023-08-18:
Can you see from the commands that were after "Install Complete" if the system was corrupted?

I have a new "stupid" problem, I can't open a new terminal in Ubuntu-18.04.  I can open terminals for 20.04 and 22.04 no issue.

PS C:\Users\peter> wsl -d Ubuntu-18.04
-bash: /home/peterd/.bashrc: line 119: syntax error near unexpected token `('
-bash: /home/peterd/.bashrc: line 119: `export PATH=~/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/mnt/c/Program Files/WindowsApps/CanonicalGroupLimited.Ubuntu18.04LTS_1804.6.4.0_x64__79rhkp1fndgsc:/mnt/c/WINDOWS/system32:/mnt/c/WINDOWS:/mnt/c/WINDOWS/System32/Wbem:/mnt/c/WINDOWS/System32/WindowsPowerShell/v1.0/:/mnt/c/WINDOWS/System32/OpenSSH/:/mnt/c/Program Files/MATLAB/R2023a/runtime/win64:/mnt/c/Program Files/MATLAB/R2023a/bin:/mnt/c/Program Files/MATLAB/R2019a/runtime/win64:/mnt/c/Program Files/MATLAB/R2019a/bin:/mnt/c/Program Files/PuTTY/:/mnt/c/Program Files/dotnet/:/mnt/c/Program Files (x86)/NTP/bin:/mnt/c/Programs_AR/WSJT24/wsjtx/bin:/mnt/c/Programs_AR/JTDXimproved/159/bin:/mnt/c/Programs_AR/WSJT_improved_PLUS/wsjtx/bin:/mnt/c/Programs_AR/WSJT_FT4NS/wsjtx/bin:/mnt/c/WSJT/wsjtx/bin:/mnt/c/Users/peter/AppData/Local/Microsoft/WindowsApps:/snap/bin'

peterdri, 2023-08-18:
I ignored the error message and typed fissure and the FISSURE dashboard came up.

cpoore1, 2023-08-18:
I’ll try WSL in the next few weeks and provide an update here if it is working. There may be factors that make the install more difficult. 

You can search for instances of .bashrc in the installer file and see what was added. Chances are it has to be modified if errors are produced or terminals won’t open.

peterdri, 2023-08-19:
I am a new user of fissure but have worked with GNUradio.    I am considering to use fissure as part of teaching communications theory at my university, since it can illustrate basic concepts using real signals.     I have run teaching labs using GNUradio and USRPs for over a decade. .  Is there a beginners guide to help sort out basic startup problems.  For example, my RTL runs fine on SDRsharp in Windows 10, but the GNURadio flowgraph using the RTL source says "no supported devices found".   Also, I find that the dashboard is often unresponsive which may be user error.   Busy students quickly get frustrated.

cpoore1, 2023-08-20:
We are in the process of filling out the documentation in the Read the Docs user manual. It is mostly empty now with only a few tidbits, but it will get there eventually. Similarly, more lessons will be constructed towards demonstrating concepts for beginners. Anyone who wants to develop their own lessons is encouraged to do so and we offer the option to integrate those lessons to help others or we can do things to make it easier for them to maintain it themselves.

Your situation is what we are trying to achieve with FISSURE, although you would certainly be pioneer at this stage. Once the Phase I roadmap items are completed this summer, we will be doing more to actively reach out to academia for those looking to demonstrate principles in signals, cybersecurity, programming, and open source software in general. We can make a discussion post outside this issue in which to elaborate further. 

Depending on the FISSURE branch, different GNU Radio mechanisms and software packages are used by default. For the RTL with the maint-3.10 branch, I would check to see if only the soapy blocks are implemented in FISSURE flow graphs (Inspection Flow Graphs would be a good place to check). If the RTL is not working for certain tools or functions, we can work together to remedy it by creating a new issue and coming up with suggestions to fix it. Chances are some version of a software package was unable to be installed or conflicted with existing software and we moved on waiting for the day when someone would speak up. However if the problem happens to be attributed to a specific virtual environment, we would have to test it out sometime to confirm which steps need to be taken to ensure it works every time.

If there are certain actions that cause the Dashboard to lock up, we would be very interested in seeing the steps to reproduce the behavior in a new issue so we can try to fix it. There are plenty of optimizations that still need to be done.