ainfosec/FISSURE #46: ***Fix*** tostring() error

Beauxrel, 2023-10-09:
Ubuntu 22.04

When this issue occurs 

File "/home/landon/.local/lib/python3.10/site-packages/scapy/arch/linux.py", line 195, in get_alias_address
    names = names.tostring()
AttributeError: 'array.array' object has no attribute 'tostring'

Replace

names = names.tostring()
-> 
names = names.tobytes()

in this file
/home/{username}/.local/lib/python3.10/site-packages/scapy/arch/linux.py

cpoore1, 2023-10-09:
Someone brought this up a few months back and I thought I put in a fix... but I don't see it there. I'll check to see if this happens on a clean install and I might have to add a warning describing the fix or force the newer version of scapy with the fix as part of the install. I'll comment here with progress soon.

Beauxrel, 2023-10-17:
Oops, closed too soon. Alright I will wait for your reply

cpoore1, 2023-10-20:
I tried the install on Ubuntu 22.04.2 and noticed a bunch of problems in the installer:
- The sed command in trackerjacker didn't take hold for fixing the tobytes() problem (I'll move this to under Scapy because it is now needed). ```sudo sed -i 's/tostring/tobytes/g' /usr/local/lib/python3.10/dist-packages/scapy/arch/linux.py```
- Metasploit didn't download.
- There's a matplotlib import error that somehow gets fixed by reinstalling it. I'll have to see what lines are breaking it.

I'll make a couple one-time exceptions in the installer script for these things when it comes to the differences between 22.04.1 and 22.04.2 and add an extra 22.04.2 option in the first GUI.

cpoore1, 2023-11-13:
I added options for Ubuntu 22.04.1, 22.04.2, and 22.04.3 to the installer. I commented out the matplotlib line because I think the default version caused problems with yellowbrick. It gets downloaded from another program down the line in "Miscellaneous Dependencies" and it no longer causes an error after commenting it out. I added the scapy fix and tested it on 22.04.2. There are still some warnings, but the FISSURE Dashboard launches with all the minimum requirements checked. The versions for everything will have to be reigned in someday but it becomes increasingly difficult as the number of included programs grows.