ainfosec/FISSURE #50: Parrot OS "externally-managed-envoroment" errors

SaturnsVoid, 2024-08-04:
Fresh install of Parrot OS Desktop, When running the installer for Parrot OS its not able to install the dependencies due to a "externally-managed-envoroment" error

cpoore1, 2024-08-04:
I haven’t tried Parrot OS in a while but I can take a look soon. You might need ‘--break-system-packages’ after all the pips. Compare it with the Raspberry Pi install commands.

cpoore1, 2024-08-04:
Looks like they updated it to be like 22.04 and debian 12 bookworm based. So if it's like version 6.1 or something like that then I will have to redo everything to match the latest versions. It should behave like the raspberry pi but there's a check in FISSURE for the operating system and that is used for determined which GNU Radio/library version to use 3.8 or 3.10. 

If you don't want to wait for a fix, there's a file in `./fissure/utils/common.py` that specifies which library to use based on OS. Looks like it is missing Parrot OS so unique identifier text from `lsb_release -d` it would have to be added to OS_3_10_KEYWORDS and whatever default terminal it uses in the variables right below that.

SaturnsVoid, 2024-08-04:
> Looks like they updated it to be like 22.04 and debian 12 bookworm based. So if it's like version 6.1 or something like that then I will have to redo everything to match the latest versions. It should behave like the raspberry pi but there's a check in FISSURE for the operating system and that is used for determining which GNU Radio/library version to use 3.8 or 3.10.
> 
> If you don't want to wait for a fix, there's a file in `./fissure/utils/common.py` that specifies which library to use based on OS. Looks like it is missing Parrot OS so unique identifier text from `lsb_release -d` would have to be added to OS_3_10_KEYWORDS and whatever default terminal it uses in the variables right below that.

Thats fine i can wait, been using basic scripts anyways. I like Parrot OS over Dragon so i have to port a bunch of stuff over anyways

cpoore1, 2024-08-16:
I updated the installer and it works for all but a handful of software tools. FISSURE was updated to use the 3.10 libraries. The menu items take about 5 seconds to populate with text after a terminal opens. Unfortunately, I'm having trouble getting my computer to interact with a USRP B210. It's not being recognized correctly so it doesn't load the firmware. The command `uhd_find_devices` and `uhd_usrp_probe` fail and causes the device to be removed from `lsusb`. The `dmesg` output doesn't look right either. If anyone runs into this, I'd like to know the solution for it.