ainfosec/FISSURE #52: Install fails on DragonOS_focalX R36 (and probably with Ubuntu22.04) with setuptools > 70

serveln, 2024-08-15:
DragonOS FocalX and also Ubuntu updates setuptools to versión 72, which produces installation errors with PyYAML, py... and ...
Changing setuptools from the installer to "sudo python3 -m pip install "setuptools<71" fix the issue.

Error:
[image]

cpoore1, 2024-08-15:
I will test it out and do a deep dive of the installer for DragonOS FocalX after I get through Parrot Security 6.1. I'll check it again on Ubuntu too. I didn't notice this issue for the last Ubuntu install I did.

cpoore1, 2024-08-16:
I'm testing out DragonOS FocalX now and there are a few problems that I see:
- Installer checks Focal instead of FocalX
- DragonOS already has several programs that can be removed from the installer: qflipper, htop, qtdesigner
- Several empty installer categories can be removed to make it look better
- fissure command needs to add the root folder to the PYTHONPATH (fixes missing fissure module errors). I'm not sure why it isn't recognizing all the __init__.py files to use the fissure module.
- pyzmq needs to be greater than 23.0.0 to use the SocketType enumeration
- It's missing the Network Certificates installation item and the command needs to change to account for the missing fissure module
- I'm still testing all the tools to see if they still install correctly
- DragonOS already has PyYAML and I don't think I still need that specific version. I'll test things to see if any errors come from a newer version.

cpoore1, 2024-08-17:
I tried to force all the setuptools to use a version less than 71 for DragonOS FocalX. It is version 72 for Ubuntu 22.04 but I didn't notice any of the errors across tools during the last install or while doing a demo of FISSURE. I will reopen this issue if I see it again. I think the install for DragonOS FocalX is good minus a couple troublesome tools that I left unchecked. Thanks for the issue and pull request.

closed 2024-08-17 by cpoore1