ainfosec/FISSURE #54: Add support for SDRPlay devices + RTL_TCP

suzieq-1, 2024-08-21:
Hi,
Could you please add support for SDRPlay devices?

- RSPdx and RSPDuo

Also RTL_TCP (most used form of connection to RTL SDR) and Spyserver (used by SDR#)

cpoore1, 2024-08-21:
Ok, I'll order some now to test. I'll search for answers but I can get a head start if you or anyone else can provide some input for me:
- Supporting software and commands to install software
- How you would like to use it? Another hardware option, GNU Radio, scripts, open tools from the menu?
- Any probe commands that you can issue from the command line to get information
- Ways to specify a unique device like the serial number that works in GNU Radio or other tools

Same deal for the servers, they might be a little trickier to integrate but I'll test them out.

suzieq-1, 2024-08-22:
Thank you for your effort.

- apt install rtl-sdr on debian for RTL SDR (rtl_tcp)
- airplay API for linux here: https://www.sdrplay.com/api/, then use SDRConnect for Airplay device
- RSP_TCP for airplay device https://www.sdrplay.com/docs/SDRplay_RSP_TCP_Server_Guide.pdf
- linux spyserver is here: https://airspy.com/download/ for RTL SDR for example in use for SDR#

Mostly the receiver is remote on a raspberry pi, either use spyserver or rtl_tcp for RTL devices, then SDR Connect or RSP TCP for airplay devices. 

The best thing would be to use it directly in FISSURE then select TCP as a device source, then select RTL_TCP, Airspy, RSP_TCP, SDR Connect, it also works this way in SDR Angel. 
Maybe a little textbox/select box on the TCP connect window, whereby you can give it a number so you can load multiple remote SDR's in.

cpoore1, 2024-09-03:
I now have the RSPduo and RSPdx. Looks like I'll only be using them only for 22.04-like operating systems. I don't think I can install the software as part of the FISSURE installer because of all the captchas. Best I can do is write the steps down and load the SDRconnect command from the FISSURE menu in the default location of where it gets installed. I might be able to do something with some of the plugins.

It doesn't seem to like the Soapy SDRPlay Source block out of the box in GNU Radio Companion. Nor does it appear to work after the API and SDRconnect software are installed. SDRconnect is working fine. I tried installing [gr-sdrplay3](https://github.com/fventuri/gr-sdrplay3) and that provides more GNU Radio blocks that work. I might not provide an option to change the antennas for the blocks since the fields are comboboxes (I have to check to see if I can tweak those settings in the FISSURE code). Gain is measured by attenuation in the block so I can either use the attenuation value or subtract it from whatever the max is to avoid doing things backwards. I'm not sure how to specify specific radios of the same type. I'm guessing it is the "RSP Selector" field and you give it a value of 0, 1, 2, etc. I only have one of each so I can't really test that other than knowing setting it to 1 produces an error for me but nothing if it is set to 0.

I'm not sure how to probe it. `SoapySDRUtil` commands don't do anything because SDRplay isn't installed as a module. I could do `lsusb`.

cpoore1, 2024-09-25:
Base support for the RSPduo, RSPdx, and RSPdx R2 using the gr-sdrplay3 blocks was added. This includes TSI detection flow graphs, IQ recording flow graphs, and IQ inspection flow graphs. New attacks should be able to be added to the library with these hardware types from the Library tab. We'll work on copying and testing existing attack scripts/flow graphs to all the radio hardware in the future. I still need to implement a good way of switching the antennas in the blocks with code, so switching is disabled for now.

I can work towards integrating the remaining SDRplay radios but I would need someone else to test them. These include RSP1, RSP1A, RSP1B, and RSP2.