ainfosec/FISSURE #56: ModuleNotFoundError: No module named 'distro'

Dzunior, 2024-09-28:
Hi,
After installing and running :
`fissure`
I'm getting:
`ModuleNotFoundError: No module named 'distro'`

But then:
```
pip3 install distro
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: distro in /usr/lib/python3/dist-packages (1.4.0)
```

Am I missing something?

cpoore1, 2024-09-28:
I think the rest of FISSURE uses a format like `sudo python3 -m pip install distro` when installing Python packages. We can likely remove the sudo part someday for them all. The install can fail sometimes if it’s formatted differently because pip3 and python3 don’t always line up. I can check the install again to see if that’s a new package we forgot.

cpoore1, 2024-09-29:
I removed the distro package. It was recently added for testing and was not being used. Thanks for pointing it out.

closed 2024-09-29 by cpoore1