ainfosec/FISSURE #60: Installation failure on Raspberry Pi 4

richtoy, 2025-01-16:
I am trying to install FISSURE on a freshly installed and updated RPi 4

uname -a
Linux Fissure 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux

The installation starts and there is a failure from which it recovers:
error: snap "netron" is not available on stable for this architecture (arm64) but exists on other
       architectures (amd64).

After a while there is another failure that is effectively fatal:
E: Unable to locate package docker-compose-v2

Once this has occurred we are in a loop:

localhost:5432 - no response
Waiting for PostgreSQL to be ready...
localhost:5432 - no response
Waiting for PostgreSQL to be ready...

I tried installing docker and docker-compose but the error remains.

There are docker-compose-v2 guides here:
[](https://www.diyenjoying.com/2022/05/12/install-docker-compose-onto-raspberry-os/)
[](https://docs.docker.com/engine/install/debian/)

Should these be followed?

cpoore1, 2025-01-16:
I’ll check the raspberry pi install this weekend and update this issue and code. Thanks for pointing out the errors.

cpoore1, 2025-01-18:
I believe I fixed the errors. I pushed the changes now. There was no snap for netron so I installed it from source with:
```
sudo apt install -y nodejs npm build-essential libgtk-3-dev libwebkit2gtk-4.0-dev
mkdir -p ~/Installed_by_FISSURE
cd ~/Installed_by_FISSURE
git clone https://github.com/lutzroeder/netron.git
cd netron
npm install
npm run electron:build
```
Then I call it with `npm start` from the netron folder. The user will have to browse to open a model until the `netron` command is restored. I had trouble making the command reappear so loading a file directly will have to wait until someone figures that out. There's also an online version of netron that can be used from a browser.

Docker was not installing either with `sudo apt-get install -y docker.io docker-compose-v2` so I replaced those lines with:
```
mkdir -p ~/Installed_by_FISSURE
cd ~/Installed_by_FISSURE/
curl -fsSL https://get.docker.com -o get-docker.sh
sudo sh get-docker.sh
sudo apt-get install -y docker-compose-plugin
sudo usermod -aG docker ${USER}  # Reboot computer to use docker commands without sudo
```

There was another error in _Frontend.py_ for Raspberry Pi OS that required "actionICE9_Bluetooth_Sniffer" to be renamed to "actionICE9_Bluetooth_Scanner" in order for the Dashboard to load up.

I also fixed an error when the database docker container is not already running when the `fissure` command is issued. After issuing the command to start the docker container it needed a sleep before connecting to it. So now it does 10 retries every two seconds by default when opening a connection to the database.

The stylesheets still look a little off in Raspberry Pi OS and there a lot of other tools that don't work. Let me know if these updates do or don't fix your problems or you see other things that are broken.

richtoy, 2025-01-19:
Thank you for you efforts.  I have now successfully built FISSURE on the RPi4 but am unable to select a sensor.  I have a HackRF One connected to the RPi4 that is correctly detected at the USB level but the pane within the dialog is empty.

[image]

I have also setup wsl on my Windows 11 machine and am attempting to build FISSURE but have encountered different issues that I will report tomorrow in a new ticket.  Which platform/configuration is the most developed/tested/reliable?

cpoore1, 2025-01-20:
If you click "Launch" (with or without recall settings on connect), it should load up bottom part of that window after a few seconds. From there you can check HackRF from the "Mass Hardware Scan" list and click "Auto Scan". If `hackrf_info` works on your system it should return the serial number. Then you click "Add to All" and "Apply". That should populate the HackRF and its serial number in all the comoboxes throughout the Dashboard. I was able to scan my HackRF on the Raspberry Pi just now. Make sure you installed the HackRF hardware item in the FISSURE installer. A simple test to see if the HackRF flow graphs work is with the IQ Data > Inspection tab. Double click an item on the left like "waterfall_hackrf.py" and click "Start".

I haven't tried FISSURE with WSL on Windows. That is something I have been meaning to look into though. However, it does feel like a lot of work would be involved. Any of the Ubuntu's >20.04 will be the most regularly tested and developed on until I can get a better automated testing system up and running. For a small form factor I'd recommend the Orange Pi (Ubuntu ARM) or a mini PC. There are a lot of tools that don't work with Raspberry Pi OS.

richtoy, 2025-01-20:
Thanks, I can confirm that its all working on the RPi, I will have a play.  I have abandoned the wsl but would I have more success with an Virtualbox Ubuntu VM?  My laptop is faily well equiped so I can throw quite some cores and memory at it.  Thanks again for your help.

richtoy, 2025-01-21:
I managed to completely install FISSURE on a 22.04 VirtualBox VM but it fails to start and dumps core.  syslog gives more info.

```
ad@Fissure:~$ fissure
/home/dad/.local/bin/fissure: line 3: 177968 Illegal instruction     (core dumped) python3 /home/dad/FISSURE/fissure/Dashboard/__main__.py

ad@Fissure:/var/log$ tail syslog 
2025-01-21T16:56:14.049453+00:00 Fissure kernel: traps: python3[177809] trap invalid opcode ip:793a673185fa sp:7ffda680b270 error:0 in libtensorflow_framework.so.2[793a66bb3000+168e000]
```

cpoore1, 2025-01-22:
This problem appears to center around Tensorflow and the VM. Somehow avx/avx2 needs to be supported for that version of Tensorflow or a different version of Tensorflow has to be installed. I can check it out in a few days because this seems like a common problem people will run into. Try `lscpu | grep avx` in the VM to see if anything shows up.

richtoy, 2025-01-22:
Thanks, lots of discussion [here ](https://forums.virtualbox.org/viewtopic.php?t=109776)

I have tried many of the internet suggestions without success.  I will experiment further tomorrow.

The OP's eventual solution was to revert to a Windows 10 Host, I will not be doing that...

richtoy, 2025-01-24:
> Thanks, lots of discussion [here ](https://forums.virtualbox.org/viewtopic.php?t=109776)
> 
> I have tried many of the internet suggestions without success. I will experiment further tomorrow.
> 
> The OP's eventual solution was to revert to a Windows 10 Host, I will not be doing that...

Well I eventually succeeded!  The important things are to make changes to Windows such that the green turtle with a V is replaced by a blue box with a V in the status line of the VirtualBox guest (see below) **AND** to set the paravitualization to minimal for the guest.

[image]

To Windows changes I made were:

Open Windows PowerShell as administrator and run the following commands to disable Hyper-V and VirtualMachinePlatform features

Disable-WindowsOptionalFeature -FeatureName VirtualMachinePlatform -Online
Disable-WindowsOptionalFeature -FeatureName Microsoft-Hyper-V-All -Online

Disable the hypervisor on your Windows system by running
bcdedit /set hypervisorlaunchtype off

open the Group Policy Editor and in folder
Computer Configuration > Administrative Templates > System > Device Guard
and set "Turn on Virtualization Based Security" to Disabled

Open Windows PowerShell as administrator and run the following command
Set-ItemProperty -Path "HKLM:\System\CurrentControlSet\Control\DeviceGuard" -Name "EnableVirtualizationBasedSecurity" -Value 0

For unknown reasons I don't think the installation of FISSURE is perfect so I will revert to an earlier snapshot and install again.

cpoore1, 2025-01-24:
I was able to get FISSURE to install and run a HackRF waterfall from the Inspection tab on WSL2 Ubuntu (Ubuntu 24.04). I ran into the same problems you did with CUDA, cuDNN, xcb, PyQt, etc. I also had to do a bunch of steps to get USB passthrough using PowerShell. There were a couple other standard packages missing in Ubuntu that I can add to the installer (things like usbtools, and unzip).

I'll try it from scratch again and write my steps down here and update the installer, user manual, and/or README.

cpoore1, 2025-01-26:
I updated the README with WSL2 instructions. I changed the TensorFlow installation to the CPU version so I'll see if that helps with some of the errors. If people want to try TensorFlow with GPUs I had to install this software from these sites for NVIDIA support that will change based on OS version:
- https://developer.nvidia.com/cuda-downloads?target_os=Linux&target_arch=x86_64&Distribution=Ubuntu&target_version=24.04&target_type=deb_network
- https://developer.nvidia.com/cudnn-downloads?target_os=Linux&target_arch=x86_64&Distribution=Ubuntu&target_version=24.04&target_type=deb_network

```
wget https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2404/x86_64/cuda-keyring_1.1-1_all.deb
sudo dpkg -i cuda-keyring_1.1-1_all.deb
sudo apt-get update
sudo apt-get -y install cuda-toolkit-12-8

wget https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2404/x86_64/cuda-keyring_1.1-1_all.deb
sudo dpkg -i cuda-keyring_1.1-1_all.deb
sudo apt-get update
sudo apt-get -y install cudnn

sudo apt install -y \
  libx11-xcb1 libxcb-util1 libxcb-xinerama0 \
  libxcb-icccm4 libxcb-image0 libxcb-keysyms1 \
  libxcb-render-util0 libxcb-shape0 libxcb-randr0 \
  libxcb-xfixes0 libxcb-sync1 libxcb-xkb1 \
  libxkbcommon0 libxkbcommon-x11-0

export QT_QPA_PLATFORM=xcb
```

I'll retest the FISSURE installer in a little bit to verify I got everything. I could always make it an option on how to install TensorFlow.

cpoore1, 2025-01-27:
I tested it a bit more and `opencv-python` causes the xcb issues with WSL2. I'm going to replace it with opencv-python-headless. If this causes future issues with certain tools, we'll know why. There are still some GPU warnings popping up that could be suppressed but I don't think are hurting anything. Ubuntu is missing `ubuntu-standard` tools and other programs like `eog` so I'll add them to the installer. Certain programs take longer to operate in WSL2. Opening an image with eog takes a couple minutes for a large image but small programs like `feh` or `display` load instantly. There are probably other cases like that throughout FISSURE.

```
sudo python3 -m pip uninstall opencv-python --break-system-packages
sudo python3 -m pip install opencv-python-headless --break-system-packages
```