ainfosec/FISSURE #61: Run fissure  on Ubuntu 24.04 error

ydfang, 2025-01-23:
Hello cpoore1, 

I installed FISSURE on Ubuntu 24.04. After installation, I ran Fissure and it had errors. 

I first made a FULL install, then I had errors when I ran fissure. 
Then I reinstalled it with Minunum, I still had errors.

The errors are: copied below. Please help.

------------------------------------------

➜  FISSURE git:(Python3) ✗ fissure  
/usr/lib/python3/dist-packages/scapy/layers/ipsec.py:512: CryptographyDeprecationWarning: TripleDES has been moved to cryptography.hazmat.decrepit.ciphers.algorithms.TripleDES and will be removed from cryptography.hazmat.primitives.ciphers.algorithms in 48.0.0.
  cipher=algorithms.TripleDES,
/usr/lib/python3/dist-packages/scapy/layers/ipsec.py:516: CryptographyDeprecationWarning: TripleDES has been moved to cryptography.hazmat.decrepit.ciphers.algorithms.TripleDES and will be removed from cryptography.hazmat.primitives.ciphers.algorithms in 48.0.0.
  cipher=algorithms.TripleDES,
2025-01-22 22:12:22.645095: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:477] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
E0000 00:00:1737601942.661799   34170 cuda_dnn.cc:8310] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
E0000 00:00:1737601942.666440   34170 cuda_blas.cc:1418] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
QObject::moveToThread: Current thread (0x3f652a00) is not the object's thread (0x473bdf10).
Cannot move to target thread (0x3f652a00)

qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/usr/local/lib/python3.12/dist-packages/cv2/qt/plugins" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb, linuxfb, minimal, offscreen, vnc, webgl.

Aborted (core dumped)

cpoore1, 2025-01-23:
It looks like your setup deviates significantly from the default environment, introducing conflicts between Python libraries, Qt plugins, and GPU components. We might be able to account for the problems if we figure out what they are. Tensorflow is causing a lot of extra problems for people that I might have to do more with someday. Here are some things to try but I can't really comment on what's going to happen if you go through with it. If you see these problems on a clean Ubuntu install then I'll try to adapt to it.

ChatGPT says for the Qt Platform Plugin Error:

Install Missing Dependencies for the xcb plugin:
`sudo apt install libx11-xcb1 libxcb1 libxcb-util1 libxcb-keysyms1 libxcb-image0 libxcb-shm0 libxcb-icccm4 libxcb-randr0 libxcb-render-util0 libxcb-xinerama0 libxcb-xfixes0`

Ensure Qt is Using the Correct Plugins Path: The error suggests the wrong path for the plugins. Try explicitly setting the QT_QPA_PLATFORM_PLUGIN_PATH environment variable to the correct directory:

`export QT_QPA_PLATFORM_PLUGIN_PATH=/usr/lib/x86_64-linux-gnu/qt5/plugins`

Add this line to your .bashrc or .zshrc if the issue persists.

Check Permissions: Ensure the plugin directory and its files have the correct permissions:
`sudo chmod -R 755 /usr/lib/x86_64-linux-gnu/qt5/plugins`

Reinstall PyQt5: If you installed Qt via pip, there might be a mismatch. Reinstall PyQt5 and its dependencies (I suggest installing PyQt5 the same way that is listed in install.sh for Ubuntu 24.04):
```
pip uninstall PyQt5 PyQt5-sip
pip install PyQt5==5.15.7
```

CUDA Errors:

Ensure Correct CUDA Version: Check that the installed version of CUDA matches the version your application and TensorFlow are expecting:
`nvcc --version`
If mismatched, reinstall the correct version.

Clear Conflicting Libraries: Ensure only one set of CUDA libraries is in use. Update your LD_LIBRARY_PATH:
`export LD_LIBRARY_PATH=/usr/local/cuda/lib64:$LD_LIBRARY_PATH`

Verify TensorFlow GPU Support: Run a quick test to confirm TensorFlow recognizes your GPU:
```
import tensorflow as tf
print(tf.config.list_physical_devices('GPU'))
```

ydfang, 2025-01-23:
I re-install Ubuntu 24.04 with Minunum Fissure installation, and it works.

Thank you.

cpoore1, 2025-01-27:
The xcb errors are associated with opencv-python (see #60). It was recently added to the list of FISSURE dependencies and this error was overlooked. If it is removed and replaced with the headless version it should open. I have updated the installer to fix this so it shouldn't happen on new installs. I'll reexamine why this package is required in the future. If it was installed without sudo for some other purpose, it can be removed without using sudo. If it is a newer operating system and the software was installed with `--break-system-packages`, then it is needed to uninstall. If it was installed without the flag, you don't need it while uninstalling. Thanks for bringing it to our attention.

```
sudo python3 -m pip uninstall opencv-python --break-system-packages
python3 -m pip uninstall opencv-python --break-system-packages
sudo python3 -m pip install opencv-python-headless --break-system-packages
```